jbots/kicad-jb-public #19: Add MIC5213

justyn, 2020-09-06:
Please check or create this part:

MIC5213

Datasheet: http://ww1.microchip.com/downloads/en/DeviceDoc/mic5213.pdf

This is an SC-70 package and it is expected that the existing footprint will be sufficient, however it will need to be checked.

Aliases: please add all the relevant aliases for this part from the datasheet. The "B" or "Y" in each part name should be replaced with an x (see [KLC rule S2.2](https://kicad-pcb.org/libraries/klc/S2.2/))

For example the part should be made for MIC5213-2.5xC5, and aliases added for MIC5213-2.6xC5, MIC5213-2.7xC5 etc.

**Stages**

This ticket will track the six stages of making the part(s).

Each time you start a new stage, please **cut and paste** the comment box for that stage into a new comment. After submitting the comment use your mouse to click the tickboxes, and use the "edit" button the comment to fill in the answers as you go. Please don't create duplicate comments.

## Stage 1: Check for symbols and footprints in official libraries

Check for existing parts in [kicad-symbols](https://github.com/KiCad/kicad-symbols) and [kicad-footprints](https://github.com/KiCad/kicad-footprints)

Cut and paste comment box for stage 1:
```
Write here any symbols for this part (or write NO_EXISTING_SYMBOL):
Write here any alternative parts that are pin-compatible and could be added as an alias (or write NO_PIN_COMPATIBLE):
Write here any footprints that could already be used (or write NO_EXISTING_FOOTPRINT): SOT-353_SC-70-5

 - [ ] For any parts written above, carefully verify against the datasheet (or tick if no parts)
```

## Stage 2: Create/modify required parts

The parts must follow all the rules of the [Kicad Library Conventions](https://kicad-pcb.org/libraries/klc/) and also be similar to existing parts wherever possible.

Some of the KLC tests can be checked on your computer using [kicad-library-utils](https://github.com/KiCad/kicad-library-utils/). Note that this doesn't cover all tests, the rest must be checked manually.

Cut and paste comment box for stage 2:
```
- [ ] Check whether the footprint could be created by modifying some values in [kicad-footprint-generator](https://github.com/pointhi/kicad-footprint-generator/) (or tick if no footprints needed)

Write here any footprints to be created/modified normally (or write NO_FOOTPRINT): NO_FOOTPRINT
Write here any footprints to created/modified with scripting (or write NO_SCRIPTING): NO_SCRIPTING

Write here any symbols to be created/modified as new parts (or write NO_SYMBOL):
Write here any symbols to be added as an alias (or write NO_ALIAS):

- [ ] Make or modify the parts
- [ ] Satisfy **all** parts of the [Kicad Library Conventions](https://kicad-pcb.org/libraries/klc/)
- [ ] Ensure that symbol is similar to similar parts already in the library, retaining identical pin spacing where possible
```

## Stage 3: Create pull request in kicad-jb-public

For the [kicad-jb-public](https://github.com/jbots/kicad-jb-public) repository, both symbol and footprint are added to the **same repository**.

The rules for where to place the parts are different to those for the official libraries:

**Symbols**:
* create a new `.lib`/`.dcm` file for the parts directly in the `sym` directory (and not in any subdirectory)
* there shouldn't be any other parts in this library
* if there is a new footprint also being added to this repository, the library should be `fp`
* name the lib by appending the part number to the usual library name from kicad-symbol
* For example: if the parts are "ISL328x" and they are designed to be added to Interface_UART.lib in kicad-symbols, they should be added to a new library at the path:
`sym/Interface_UART_ISL328x.lib` (and `.dcm`)

**Footprints**:
* create a new `.kicad_mod` file directly in the `fp.pretty` directory (and not in any subdirectory)
* For example: if the package is "QFN-28_4x4mm_P0.5mm" it should be added at the path:
`fp.pretty/QFN-28_4x4mm_P0.5mm.kicad_mod`

Create a pull request (PR) in [kicad-jb-public](https://github.com/jbots/kicad-jb-public) (this repository) with the part number in the title. Make sure your branch is up to date with the repository first.

Cut and paste comment box for stage 3:
```
- [ ] Create a new branch in your fork of this repository (don't work on "main" branch)
Link to pull request in this repository:
```

## Stage 4: Make any changes to allow merging into this repository

Automatic checks will be run when you create your pull request in this repository.

The parts have not passed until you see "Check symbol _name_" or "Check footprint _name_" next to the name of **each** symbol or footprint you have created, with a green tick.

The Kicad Library Conventions can be found [here](https://kicad-pcb.org/libraries/klc/).

Any KLC rule failures must be fixed. Click on "Details" next to the failing rule for more information. If the automatic check should be ignored in this specific case, explain it below.

**Important**: Not all rules have an automatic test, so a part passing does **not** mean that it doesn't violate any rules - they must be read carefully.

Cut and paste comment box for stage 4:
```
Explanation of any failing checks that should be ignored (or write NO_IGNORE):
- [ ] All symbols and footprints are listed as passing, or failing checks explained
```

## Stage 5: Create pull requests in kicad official libraries

**Stages 5 and 6 skipped while kicad-symbols repository is in lockdown (September 2020)**

```

SagarSSM, 2020-09-06:
Write here any symbols for this part (or write NO_EXISTING_SYMBOL):
Write here any alternative parts that are pin-compatible and could be added as an alias (or write NO_PIN_COMPATIBLE):
Write here any footprints that could already be used (or write NO_EXISTING_FOOTPRINT): SOT-353_SC-70-5

 - [ ] For any parts written above, carefully verify against the datasheet (or tick if no parts)

SagarSSM, 2020-09-06:
- [ ] Check whether the footprint could be created by modifying some values in [kicad-footprint-generator](https://github.com/pointhi/kicad-footprint-generator/) (or tick if no footprints needed)

Write here any footprints to be created/modified normally (or write NO_FOOTPRINT): NO_FOOTPRINT
Write here any footprints to created/modified with scripting (or write NO_SCRIPTING): NO_SCRIPTING

Write here any symbols to be created/modified as new parts (or write NO_SYMBOL):
Write here any symbols to be added as an alias (or write NO_ALIAS):

- [ ] Make or modify the parts
- [ ] Satisfy **all** parts of the [Kicad Library Conventions](https://kicad-pcb.org/libraries/klc/)
- [ ] Ensure that symbol is similar to similar parts already in the library, retaining identical pin spacing where possible

SagarSSM, 2020-09-06:
- [ ] Create a new branch in your fork of this repository (don't work on "main" branch)
Link to pull request in this repository:

SagarSSM, 2020-09-06:
Explanation of any failing checks that should be ignored (or write NO_IGNORE):
- [ ] All symbols and footprints are listed as passing, or failing checks explained

justyn, 2020-09-10:
Added to repository, can't be added to kicad-symbols yet.

closed 2020-09-10 by justyn