jbots/kicad-jb-public #26: Add Osram SideLED 3010

justyn, 2020-09-15:
Please check or create this footprint:

Osram SideLED 3010 from https://dammedia.osram.info/media/resource/hires/osram-dam-4861654/LB%20Y8SG_EN.pdf

**Stages**

This ticket will track the six stages of making the part(s).

Each time you start a new stage, please **cut and paste** the comment box for that stage into a new comment. After submitting the comment use your mouse to click the tickboxes, and use the "edit" button the comment to fill in the answers as you go. Please don't create duplicate comments.

## Stage 1: Check for symbols and footprints in official libraries

Check for existing parts in [kicad-footprints](https://github.com/KiCad/kicad-footprints)

Cut and paste comment box for stage 1:
```
Write here any footprints that could already be used (or write NO_EXISTING_FOOTPRINT):

 - [ ] For any parts written above, carefully verify against the datasheet (or tick if no parts)
```

## Stage 2: Create/modify required parts

Cut and paste comment box for stage 2:
```
- [ ] Make or modify the parts
- [ ] Satisfy **all** parts of the [Kicad Library Conventions](https://kicad-pcb.org/libraries/klc/)
```

## Stage 3: Create pull request in kicad-jb-public

**Footprints**:
* create a new `.kicad_mod` file directly in the `JB.pretty` directory (and not in any subdirectory)
* For example: if the package is "QFN-28_4x4mm_P0.5mm" it should be added at the path:
`fp.pretty/QFN-28_4x4mm_P0.5mm.kicad_mod`

Create a pull request (PR) in [kicad-jb-public](https://github.com/jbots/kicad-jb-public) (this repository) with the part number in the title. Make sure your branch is up to date with the repository first.

Cut and paste comment box for stage 3:
```
- [ ] Create a new branch in your fork of this repository (don't work on "main" branch)
Link to pull request in this repository:
```

## Stage 4: Make any changes to allow merging into this repository

Automatic checks will be run when you create your pull request in this repository.

The parts have not passed until you see "Check footprint _name_" next to the name of each footprint you have created, with a green tick.

The Kicad Library Conventions can be found [here](https://kicad-pcb.org/libraries/klc/).

Any KLC rule failures must be fixed. Click on "Details" next to the failing rule for more information. If the automatic check should be ignored in this specific case, explain it below.

**Important**: Not all rules have an automatic test, so a part passing does not mean that it doesn't violate any rules.

Cut and paste comment box for stage 4:
```
Explanation of any failing checks that should be ignored (or write NO_IGNORE):
- [ ] All symbols and footprints are listed as passing, or failing checks explained
```

## Stage 5: Create pull requests in kicad official libraries

Create a pull request in [kicad-footprints](https://github.com/KiCad/kicad-footprints)

Cut and paste comment box for stage 5:
```
**Before** you create a pull request(s) tick off all the following

- [ ] Create a new branch in your fork of kicad-footprints (don't work on "master" branch)
- [ ] Make sure your branch is up to date
- [ ] Place the part in the correct `.pretty` directory (for a footprint)
- [ ] Create screenshots of your parts for your Pull Requests [like this](https://kicad-pcb.org/img/klc/adding_screenshot.png).

Then:

- [ ] Create pull request using template text, filling in the details and adding screenshots
- [ ] Add link to pull request for footprint in the pull request for the symbol, and vice versa

Link to footprint pull request (or NO_FOOTPRINT_PR):
```

## Stage 6: Make required changes for parts to be merged in official libraries

At the bottom of the PR in kicad-footprints, click "Details" next to "continuous-integration/travis-ci/pr" to see any automatic recommendations for fixes to the parts.

Make any changes requested by the Kicad librarians.

Do not request a review, or send comments asking for your part to be checked. If nobody is checking the part, add more details to the pull request, including screenshots from the datasheet (of pin tables, package dimensions etc).

Cut and paste comment box for stage 6:
```
**Each time you push changes to the pull request**:
- [ ] Make any changes required by the Kicad librarians to make the part suitable for inclusion in the official libraries.
- [ ] Update the part screenshots in the **same place as the originals** by editing your comment (don't create a new comment)
- [ ] Update the PR in [kicad-jb-public](https://github.com/jbots/kicad-jb-public) (this repository) along with a comment explaining what has changed
```

dudelectric, 2020-09-18:
**Stage1**

NO_EXISTING_FOOTPRINT

 - [] For any parts written above, carefully verify against the datasheet (or tick if no parts)

dudelectric, 2020-09-18:
stage 2:

- [ ] Make or modify the parts
- [ ] Satisfy **all** parts of the [Kicad Library Conventions](https://kicad-pcb.org/libraries/klc/)

dudelectric, 2020-09-18:
stage 3:
PR link: https://github.com/jbots/kicad-jb-public/pull/27
- [x] Create a new branch in your fork of this repository (don't work on "main" branch)
Link to pull request in this repository:

dudelectric, 2020-09-18:
stage 4:

Explanation of any failing checks that should be ignored (or write NO_IGNORE):
- [] All symbols and footprints are listed as passing, or failing checks explained

dudelectric, 2020-09-18:
stage 5:

- [ ] Create a new branch in your fork of kicad-footprints (don't work on "master" branch)
- [ ] Make sure your branch is up to date
- [ ] Place the part in the correct `.pretty` directory (for a footprint)
- [ ] Create screenshots of your parts for your Pull Requests [like this](https://kicad-pcb.org/img/klc/adding_screenshot.png).

Then:

- [ ] Create pull request using template text, filling in the details and adding screenshots
- [ ] Add link to pull request for footprint in the pull request for the symbol, and vice versa

Link to footprint pull request (or NO_FOOTPRINT_PR):

dudelectric, 2020-09-21:
stage 6:

**Each time you push changes to the pull request**:
- [ ] Make any changes required by the Kicad librarians to make the part suitable for inclusion in the official libraries.
- [ ] Update the part screenshots in the **same place as the originals** by editing your comment (don't create a new comment)
- [ ] Update the PR in [kicad-jb-public](https://github.com/jbots/kicad-jb-public) (this repository) along with a comment explaining what has changed