jbots/kicad-jb-public #28: Add AM26LV31EIRGY

justyn, 2020-10-08:
Please check or create this part:

AM26LV31EIRGY (VQFN package)

Each time you start a new stage, please **cut and paste** the comment box for that stage into a new comment. After submitting the comment use your mouse to click the tickboxes, and use the "edit" button the comment to fill in the answers as you go. Please don't create duplicate comments.

## Stage 1: Check for symbols and footprints in official libraries

Check for existing parts in [kicad-symbols](https://github.com/KiCad/kicad-symbols) and [kicad-footprints](https://github.com/KiCad/kicad-footprints)

Cut and paste comment box for stage 1:
```
Write here any symbols for this part (or write NO_EXISTING_SYMBOL):
Write here any alternative parts that are pin-compatible and could be added as an alias (or write NO_PIN_COMPATIBLE):
Write here any footprints that could already be used (or write NO_EXISTING_FOOTPRINT):

 - [ ] For any parts written above, carefully verify against the datasheet (or tick if no parts)
```

## Stage 2: Create/modify required parts

Cut and paste comment box for stage 2:
```
- [ ] Check whether the footprint could be created by modifying some values in [kicad-footprint-generator](https://github.com/pointhi/kicad-footprint-generator/) (or tick if no footprints needed)

Write here any footprints to be created/modified normally (or write NO_FOOTPRINT):
Write here any footprints to created/modified with scripting (or write NO_SCRIPTING):

Write here any symbols to be created/modified as new parts (or write NO_SYMBOL):
Write here any symbols to be added as an alias (or write NO_ALIAS):

- [ ] Make or modify the parts
- [ ] Satisfy **all** parts of the [Kicad Library Conventions](https://kicad-pcb.org/libraries/klc/)
```

## Stage 3: Create pull request in kicad-jb-public

For the [kicad-jb-public](https://github.com/jbots/kicad-jb-public) repository, both symbol and footprint are added to the **same repository**.

The rules for where to place the parts are different to those for the official libraries:

**Symbols**:
* create a new `.lib`/`.dcm` file for the parts directly in the `sym` directory (and not in any subdirectory)
* there shouldn't be any other parts in this library
* name the lib by appending the part number to the usual library name from kicad-symbol
* For example: if the parts are "ISL328x" and they are designed to be added to Interface_UART.lib in kicad-symbols, they should be added to a new library at the path:
`sym/Interface_UART_ISL328x.lib` (and `.dcm`)

**Footprints**:
* create a new `.kicad_mod` file directly in the `fp.pretty` directory (and not in any subdirectory)
* For example: if the package is "QFN-28_4x4mm_P0.5mm" it should be added at the path:
`fp.pretty/QFN-28_4x4mm_P0.5mm.kicad_mod`

Create a pull request (PR) in [kicad-jb-public](https://github.com/jbots/kicad-jb-public) (this repository) with the part number in the title. Make sure your branch is up to date with the repository first.

Cut and paste comment box for stage 3:
```
- [ ] Create a new branch in your fork of this repository (don't work on "main" branch)
Link to pull request in this repository:
```

## Stage 4: Make any changes to allow merging into this repository

Automatic checks will be run when you create your pull request in this repository.

The parts have not passed until you see "Check symbol _name_" or "Check footprint _name_" next to the name of **each** symbol or footprint you have created, with a green tick.

The Kicad Library Conventions can be found [here](https://kicad-pcb.org/libraries/klc/).

Any KLC rule failures must be fixed. Click on "Details" next to the failing rule for more information. If the automatic check should be ignored in this specific case, explain it below.

**Important**: Not all rules have an automatic test, so a part passing does **not** mean that it doesn't violate any rules - they must be read carefully.

Cut and paste comment box for stage 4:
```
Explanation of any failing checks that should be ignored (or write NO_IGNORE):
- [ ] All symbols and footprints are listed as passing, or failing checks explained
```

UsairemAlamgeer, 2020-10-09:
**Stage: 1**

NO_EXISTING_SYMBOL
NO_PIN_COMPATIBLE
TSSOP-16_4.4x5mm_p0.65mm

 - [x] For any parts written above, carefully verify against the datasheet

**Stage: 2**

**Footprints created:**
Texas_R-PVQFN-N16_4x3.5mm_P0.5mm
Texas_R-PVQFN-N16_4x3.5mm_P0.5mm_ThermalVias
**Symbols created:**
AM26LV31EIRGY
AM26LV31EIPW
NO_ALIAS

- [x] Make or modify the parts
- [x] Satisfy **all** parts of the [Kicad Library Conventions](https://kicad-pcb.org/libraries/klc/)

**Stage: 3**

 [x] Create a new branch in your fork of this repository (don't work on "main" branch)
Link to pull request in this repository:
https://github.com/jbots/kicad-jb-public/pull/29

Explanation of any failing checks that should be ignored (or write NO_IGNORE):
- [x] All symbols and footprints are listed as passing

justyn, 2020-10-09:
Merged with b9ad77f

closed 2020-10-09 by justyn